GeekMagicClock/smalltv-ultra #1: I found that need to improve while using smallTV-Ultra.

helloooo082, 2023-10-13:
I found that need to improve while using smallTV-Ultra. 
1. When rebooting, the LCD brightness is change. 
2. The weather forcast data is not displayed when rebooting. 
Of course, I can go to the settings after rebooting, the LCD brightness and weather forcast, but there is an inconvenience, so please review the modification.

GeekMagicClock, 2023-10-14:
Hello, thank you for the feedback 
1. The brightness of LCD if initiallised to the default brightness when rebooting, because someone may forget they have set to the lowest brightness, they may thought it is damaged. After boot done, it will recover to the user set brightness, hope this will not bother you.
2. Do you mean weather forecast is not displayed after boot? Is the weather today info  correct?

helloooo082, 2023-10-14:
Hello. Thank you for your reply. 

1. I agree with you about LCD brightness. However, even after the boot is complete, the lcd brightness does not recover to the set value, but it is still the default value. 
2. I attach a picture for the weather forecast. It's a normal picture and a picture when I booted.

[image]
[image]

GeekMagicClock, 2023-10-15:
1. OK, I'll check it out.
2 The weather forecast update depend on your network, how long does it takes to get the correct data? Or can you try a different network?

Regards

helloooo082, 2023-10-15:
Hello, 
2. Weather update interval is the default setting of 20 minutes.(not changed). and I connected another network as you said, but the result is the same. I checked other networks using smartphone hotspots and tested SmallTV-Ultra near my phone.
Thanks.

GeekMagicClock, 2023-10-16:
Hi, 
2. How long does it take to get the data? Does the weather today info get instantly? Weather forecast is updated after weather today's info.

helloooo082, 2023-10-17:
Hi,
2. Today's weather, the first page, shows information immediately after booting. But the second page, Weather Forecast, is not updated and the information is marked as 0. This does not update after 1 hour and is marked as 0. However, if i change to another page and go back to the weather forecast, i get normal information.

You can easily check it if you set the weather forecast screen and reboot it. Or maybe you didn't have a problem?
Thanks.

GeekMagicClock, 2023-10-18:
> Hi, 2. Today's weather, the first page, shows information immediately after booting. But the second page, Weather Forecast, is not updated and the information is marked as 0. This does not update [...]

Yes, I don't have this kind of problem, can you leave your email, there're some updates under testing.

helloooo082, 2023-10-18:
Hi,
I send you my e-mail address. And I bought another Smalltv-ultra, which has
the same problem. You don't have a problem, but I don't understand that
only I have a problem like this.
Anyway, I think you look great trying to solve my problem. Thank you.

GeekMagicClock, 2023-10-18:
Hi, where can i get your mail? or you could send me via ifengchao1314(at)gmail.com

GeekMagicClock, 2024-03-27:
The forecast theme has been updated, please try the latest v9.0.17

GeekMagicClock, 2024-04-03:
> Hi, just got the -ultra today, very nice device! =)
> 
> I too got the issue with the 'empty' forecast after a reboot (firmware Ultra-V9.0.18) It happens when themes get switched automatically, namely between 1. and 2. My unprofessional guess is that the forecast data is not preserved between theme changes after a reboot.

Hi, yes, the data is directly updated from the internet after switch, seems this should be fixed, I'll try next update, thank you for the report~

closed 2024-05-26 by GeekMagicClock