GeekMagicClock/smalltv-ultra #3: Make the smallTV-Ultra work fully offline

johnred358, 2023-11-28:
Since the source code is not (yet) available for the public, it would be nice to be able to use the smalltv in completely offline mode, perhaps limiting its use only to clock and/or image viewer (no weather).

Right now it seems impossible to even see slideshow if the network connection is not present (even if the images are saved locally). 

Thanks in advance

GeekMagicClock, 2024-01-03:
Hi, yes, this feature could be added, but we want to keep it simple.
I have think about to add it this way, but maybe a bit complicated:
1) If people have set it to image viewer mode, the next reboot will enter into the image viewer mode even if wifi connection failed.
2) The AP mode will be started for 1 minute, and will be off if no connections.(In case people want to change the settings again, and also for security reason)

johnred358, 2024-01-08:
Those are both great ideas. 
May i also suggest the clock feature? I thing that while it's powered on it could work as a simple clock without internet access. Or it could be a problem?

johnred358, 2024-01-08:
I'm sorry i closed the issue, my bad!

Those are both great ideas. 
May i also suggest the clock feature? I thing that while it's powered on it could work as a simple clock without internet access. Or it could be a problem?

closed 2024-01-08 by johnred358

GeekMagicClock, 2024-01-29:
There's no RTC and battery inside which makes it not possible a clock without internet.

pilipovicn, 2025-12-10:
Hello, sorry for reopening this, but I think this feature would be sort of useful.

I have a local NTP server, and so it would work. 
I see the device tries connecting to 2 IPs at start (plus my local NTP), and if it fails (which it does since I blocked all my IoT devices WAN access), it restarts.
I don't see a point in the restart, I think a better idea would be to keep retrying connections periodically. 
That way it would work as usual, but if someone wants to make it work offline, they can, without restarts every 30 seconds.