GeekMagicClock/smalltv-ultra #7: Improve the 'time style' themes

dyfusive, 2023-11-29:
I have smalltv-ultra v9.0.8

Time Style 2 does not obey the date format setting. It is permanently set to YYYY/MM/DD, and does not change if I set it to DD/MM/YYYY or something else.

Also, is it possible for the option to have the seconds on the Time Style 1, Time Style 2 and Time Style 3 themes? The seconds only is available on the Weather Clock Today theme.

GeekMagicClock, 2024-01-03:
Hello, the date format is fixed in the simple time style2 in version v9.0.13, thank you for the feedback.

And I can't find a place on the screen to hold the seconds without mess the layout of the display, can you draw the place of seconds?

dyfusive, 2024-01-04:
Thanks for the fix for time style2 @GeekMagicClock.

Adding the seconds will only look good on time style2 and time style3. Is it possible to create a new time style that has the seconds something like the attached image?

[image]