GeekMagicClock/smalltv-ultra #9: Forcast screen is broken

19201003080114, 2024-02-04:
In weather forecast screen:
- Time display at the top is broken when minute is between :00 to :09 (In picture time is 12:09)
- Third day forecast at the bottom is not displayed

[image]

19201003080114, 2024-02-04:
Example city is Toronto, Canada (6167865)

GeekMagicClock, 2024-02-22:
Hello, is that the latest firmware ?

amomesso, 2024-02-22:
> Hello, is that the latest firmware ?

No, I bought it in November/December. Is there a way to find out what version is currently installed?

19201003080114, 2024-02-22:
I use latest firmware. V9.0.15

netstx, 2024-03-02:
> Hello, is that the latest firmware ?

I have the same issue and I'm using latest firmware.

[image]

Thanks!

19201003080114, 2024-03-02:
Ultra latest is V9.0.15 in GitHub

[image]

amomesso, 2024-03-04:
I updated to the last version but it does not fix the bug.
Also i don't know how @netstx can see firmware version. 
[image]

DruidSmith, 2024-03-12:
Same issue, I am also on 9.0.16.  Forecast only shows next two days and does not apply the units setting (values only come up as celsius).

It would be nice if the developer posted the source code so that others could help contribute.

jaaxx, 2024-03-16:
Can confirm.  Third day has null info and the units (C vs F) do not respect the settings.

ferrellsl, 2024-03-21:
I have the same problem under 9.0.16.

GeekMagicClock, 2024-03-26:
Thank you all for this report, please update to 9.0.17, this bug has been fixed, and the Min and Max temperature of the day has been fixed too!

ferrellsl, 2024-03-26:
> Thank you all for this report, please update to 9.0.17, this bug has been fixed, and the Min and Max temperature of the day has been fixed too!

Thanks for the fix/update.  I have three SmallTV-Ultras and confirm that this fix works on all three devices.  For others applying this update, the forecast screen will update after a few minutes.  Just apply the update and wait 20 minutes for the forecast screen to update.

ferrellsl, 2024-03-26:
There's still the issue of the forecast screen not showing Fahrenheit temperatures but I'm still grateful for the progress.  Thank you for creating such a cool little weather station.

GeekMagicClock, 2024-03-31:
Please update to V9.0.18, thank you all~

closed 2024-03-31 by GeekMagicClock

ferrellsl, 2024-03-31:
> Please update to V9.0.18, thank you all~

Confirmed, working properly on all 3 of my devices Fahrenheit temperatures work properly now on the forecast screen.  Thank you for your hard work and for making such a cool little weather station.