GeekMagicClock/smalltv-ultra #15: Reboot issue with V9.0.17

19201003080114, 2024-03-26:
After updating to V9.0.17, now I have reboot issues that I didn't have before when I had V9.0.15

Note: I have my own weather API key so my update interval is every 5 mins

I know it rebooted because:

- Forecast screen is blank with no data. Then I know it rebooted within the last 5 mins.

- I see display showing GIFTV asking to connect 192.168.4.1. I know this could happen if power is cut 3 times while booting up, or it cannot connect to wifi. But my wifi is OK (Router didn't reboot, or lost connection), and cable has not been touched or changed as when I was using V9.0.15.

Disappointing but I will downgrade back to V9.0.15. Forecast screen is broken but it is not as bad as random reboots.

GeekMagicClock, 2024-03-27:
Thank you for the report, I'll try your steps to confirm the problem.

GeekMagicClock, 2024-03-27:
Hi, what is your location? I have try set my update interval to 2mins, and the screen stays at forecast theme, 30 mins past without reboot.

19201003080114, 2024-03-27:
> Hi, what is your location? I have try set my update interval to 2mins, and the screen stays at forecast theme, 30 mins past without reboot.

Location is Toronto, Canada. 6167865

GeekMagicClock, 2024-03-27:
> > Hi, what is your location? I have try set my update interval to 2mins, and the screen stays at forecast theme, 30 mins past without reboot.
> 
> Location is Toronto, Canada. 6167865 Also happens for North York, Canada. 6091104

OK, have you set the auto switch between themes?

GeekMagicClock, 2024-03-27:
And yesterday I have updated two different v9.0.17, I doubt if you have update the earlier one, can you try the one now in the repo ?

19201003080114, 2024-03-27:
> > > Hi, what is your location? I have try set my update interval to 2mins, and the screen stays at forecast theme, 30 mins past without reboot.
> > 
> > 
> > Location is Toronto, Canada. 6167865 Also happens for North York, Canada. 6091104
> 
> OK, have you set the auto switch between themes?

Yes I have auto switch themes on
[image]

19201003080114, 2024-03-27:
> And yesterday I have updated two different v9.0.17, I doubt if you have update the earlier one, can you try the one now in the repo ?

I just download V9.0.17 again. Is the MD5 cf8da165ca6c1a42bbd441b34a74cd9c?

GeekMagicClock, 2024-03-27:
Got it, since the ram is quite limited, maybe the auto switch theme caused the problem.
Yes, MD5 correct.

GeekMagicClock, 2024-03-27:
> > And yesterday I have updated two different v9.0.17, I doubt if you have update the earlier one, can you try the one now in the repo ?
> 
> I just download V9.0.17 again. Is the MD5 cf8da165ca6c1a42bbd441b34a74cd9c? I updated it just now and I will be monitoring it.

I can reproduce that, the loop between themes would randomly cause the reboot, I'm checking.

angrycatmeowmeow, 2024-03-27:
Can confirm seeing the same behavior. I only have two themes switching.  9.0.17.
[image]

GeekMagicClock, 2024-03-28:
@angrycatmeowmeow  @19201003080114  Please try the V9.0.18 version if that's convient for you, I have tested it for one night with the default interval of 20min without reboot, I'll test it with higher frequency and report here.

19201003080114, 2024-03-28:
> @angrycatmeowmeow @19201003080114 Please try the V9.0.18 version if that's convient for you, I have tested it for one night with the default interval of 20min without reboot, I'll test it with higher frequency and report here.

I downloaded V9.0.18 and I am testing now.
Weather update interval shortest is 5 mins with own API key. If less than 5 mins, the value is not saved, and it will revert back to 5 mins (maybe it is a bug?)

angrycatmeowmeow, 2024-03-28:
> @angrycatmeowmeow @19201003080114 Please try the V9.0.18 version if that's convient for you, I have tested it for one night with the default interval of 20min without reboot, I'll test it with higher frequency and report here.

So far so good and the forecast screen Fahrenheit is fixed too. Thank you!

19201003080114, 2024-03-29:
> I downloaded V9.0.18 and I am testing now. Weather update interval shortest is 5 mins with own API key. If less than 5 mins, the value is not saved, and it will revert back to 5 mins (maybe it is a [...]

I left V9.0.18 for about 12 hours now.
No signs of rebooting with the same auto theme rotation as V9.0.15, and with weather update interval of 5 mins. 
Forecast screen did not lose data (meaning no reboot within last 5 mins), and no random appearance of GIFTV screen that requires a manual restart.
I will keep monitoring. So far it is looking good.

GeekMagicClock, 2024-03-31:
Thank you guys for the report.

Updates: It may will not reboot, but I noticed sometimes the gif under the weather screen will not be displayed, but will be recover after several loops, that is because of not enough space ram.

And another thing about the update interval limit, I'll check that, if that is a bug, will be fixed in next updates.

closed 2024-05-26 by GeekMagicClock