GeekMagicClock/smalltv-ultra #20: The image is not loading

alexisfgh, 2024-03-31:
The image doesn't load when selecting to display it. I have updated the Ultra-V9.0.18 Firmware and tried to revert to version 15, but the same error persists. The error arose when trying to upload a new image, but the TV froze and restarted, resulting in errors with the images.

[image]
[image]

[image]

GeekMagicClock, 2024-03-31:
Hello, how about loading a small jpg file ?  It will display nothing  when there's no files.

alexisfgh, 2024-03-31:
When a small jpg is loaded, there is no issue; the problem only arises with larger images.

GeekMagicClock, 2024-04-01:
Hi, there may be some reasons.
1) Please try with diffrent power or cable.
2) Try smaller file size.

Can you send that file to me to test?

alexisfgh, 2024-04-04:
When attempting to reload an image, it only subtracts storage space but does not display the loaded images, and the error message continues to appear. I've also tried using another power cable, but the same error persists.

I'm sharing a GIF image that I used.

[image]

GeekMagicClock, 2024-04-06:
If it only subtracts storage space but does not display, that maybe related to the format of gif files, you can upload it to ezgif.com and try optimize it.  I have tried the gif you uploaded, it works here.
Except the photo album theme, does the weather and time display correctly?