GeekMagicClock/smalltv-ultra #22: DST(Daylight Saving Time) not working

bauerpawel, 2024-04-01:
[image]

The time displayed is 11:00 and it is currently 10:00
DST(Daylight Saving Time) is on or off, there is no difference in the time displayed.

GeekMagicClock, 2024-04-01:
Hi, you can try disable that DST option, and reboot to check.

Mugsausages, 2024-04-01:
Have the same issue, enabling/disabling DST will work temporarily, adding an hour to the clock and then clock will go back and hour.

19201003080114, 2024-04-01:
> Have the same issue, enabling/disabling DST will work temporarily, adding an hour to the clock and then clock will go back and hour to non DST time

Try different time server. 
I have DST enabled and it worked fine.

Mugsausages, 2024-04-01:
> > Have the same issue, enabling/disabling DST will work temporarily, adding an hour to the clock and then clock will go back and hour to non DST time
> 
> Try different time server. I have DST enabled and it worked fine.

Can you recommend a server :)

19201003080114, 2024-04-01:
https://tf.nist.gov/tf-cgi/servers.cgi

Mugsausages, 2024-04-01:
> https://tf.nist.gov/tf-cgi/servers.cgi

Hero, it worked. Thanks :D

GeekMagicClock, 2024-04-02:
Hi, what is the city code or name? I can test.

bauerpawel, 2024-04-02:
city code: 3092856
City: Luboń

coolguru, 2024-04-02:
Same Here

city code: 5003369
City: New Baltimore

GeekMagicClock, 2024-04-03:
Hi **Just do not enable the DST option,** that will work, please check again.
I have checked both city, the time is correct without enable DST, with the V9.0.18.
Maybe that option should be removed in future update I guess.
[image]

[image]