GeekMagicClock/smalltv-ultra #26: Cannot revert back from customized weather API key

zpan45, 2024-04-03:
Repro steps:

1. set weather API key
2. key does not work, screen shows "Err key!"
3. set weather API key to blank and hit save
4. screen still shows "Err key!" with no weather info updated
5. go to a different tab on web portal, then come back to "Weather"
6. customized weather API key is still present

I had to do a factory reset and go through the setup again for it to work.

GeekMagicClock, 2024-04-05:
Hi, your own new registered weather key  will take hours to take effect.

Ashwagada, 2024-04-12:
> Hi, your own new registered weather key will take hours to take effect.

If I understand correctly, it was about the fact that it is not possible to delete your own API key. The only option is factory settings. Are you able to test it yourself?

GeekMagicClock, 2024-05-26:
> > Hi, your own new registered weather key will take hours to take effect.
> 
> If I understand correctly, it was about the fact that it is not possible to delete your own API key. The only option is factory settings. Are you able to test it yourself?

Yes, our api key can't be deleted, but yours could be deleted by reset or set a wrong one.

closed 2024-05-26 by GeekMagicClock