GeekMagicClock/smalltv-ultra #27: Correct time??

herby1620, 2024-04-04:
I just updated to the latest version.  The problem is that while the weather is correct for where I live, the time is one time zone to the east (even after I ticked the DST box.  From appearances the time zone is based on the weather location as there is no place to put it in the configuration pages.  From my point of view having the time zone as a separate entry (along with DST dates, day of week, week of month) would be a better way to do this, or at least an alternative to the present way of how it works.

Other than that...
It works quite well.

GeekMagicClock, 2024-04-05:
Hi, recently a lot of report of incorrect time, early or later 1 hour, the solution is DO NOT ENABLE the DST option.
But some one may need enable that option in some locations.

Yes, the timezone is binded to the weather location, our mean is to make it simpler, users do not need to set the timezone.
But now seems we have to add a timezone option, I'll see what to do.

19201003080114, 2024-04-05:
Time is still one hour off even with DST disabled 
[image]

GeekMagicClock, 2024-04-06:
> I think there could be an issue with saving settings with v9.0.18. I disabled DST, save, restart....Time is still wrong. I check DST settings, it was still enabled. I disable DST again, save, refresh page to ensure settings stay, restart....now the time is correct

Hi, I checked the DST saving process and found no problems and it tested fine on my device. If you can reproduce this issue please let me know, thanks。

Ashwagada, 2024-04-12:
> Hi, recently a lot of report of incorrect time, early or later 1 hour, the solution is DO NOT ENABLE the DST option. But some one may need enable that option in some locations.
>
> Yes, the timezone is [...]

It's worth adding a time zone.

ap0ught, 2024-04-22:
I am also having the issue with 5089178 right now it is 9:30 if I add dst it will give me 11:30 without it will give me 10:30

GeekMagicClock, 2024-04-22:
> I am also having the issue with 5089178 right now it is 9:30 if I add dst it will give me 11:30 without it will give me 10:30

Hi, please try disable the DST option, and restart it.

JoeB2, 2024-04-25:
> I just updated to the latest version. The problem is that while the weather is correct for where I live, the time is one time zone to the east (even after I ticked the DST box). From appearances the [...]

Same experience: My Weather location is correct: 5786882 but the time is +1hr.  Checking and unchecking the DST box followed by Save has no effect, i.e. the time remains +1hr.  Please Note: When first changing the weather location the device momentarily displays the correct time.  Following the first auto update (assume weather update) the time goes to +1hr of the correct time.

closed 2024-05-11 by GeekMagicClock