GeekMagicClock/smalltv-ultra #28: Loop Gif

mawar108, 2024-04-04:
I do see a setting to let looping gifs, so it would be nice if there would be a option to do that :D because now a gif what is only "5 sec" long and the display time is longer it would be stop animating after 5 secs or is it a bug ? ^^"

GeekMagicClock, 2024-04-05:
OK, I'll check it out, thanks for reporting~

GeekMagicClock, 2024-04-06:
I reproduced this problem when setting up a loop between themes. If you only have one gif file, it is recommended that you "set" the gif, and it will continue to play the gif file in a loop without stopping.

mawar108, 2024-04-06:
i thought so... but isnt it possible to let it loop even with this combination of settings?

GeekMagicClock, 2024-04-09:
> i thought so... but isnt it possible to let it loop even with this combination of settings?

Not available for now.

Ashwagada, 2024-04-12:
> I dont see a setting to let looping gifs, so it would be nice if there would be a option to do that :D because now a gif what is only "5 sec" long and the display time is longer it would be stop animating after 5 secs or is it a bug ? ^^"

Good idea.

GeekMagicClock, 2024-04-20:
Hi, please update to v9.0.20, if the play interval in the album theme is longer than the loop theme interval, the gif will still be played, you can try and please let me know, thank you.

GeekMagicClock, 2024-05-26:
Reopen if needed.

closed 2024-05-26 by GeekMagicClock