GeekMagicClock/smalltv-ultra #29: Display artifact with V9.0.18

19201003080114, 2024-04-05:
City: North York, Canada
City code: 6091104
Firmware: V9.0.18

[image]

TrueLoonix, 2024-04-06:
Your location gives no artifacts on my device.
Does it give the same glitch in other locations?
Did you try to re-flash your ultra?

https://github.com/GeekMagicClock/smalltv-ultra/assets/92674519/24f1e2da-dcd9-4d58-8513-4ce07bd42abc

GeekMagicClock, 2024-04-09:
Hi, @19201003080114  that would happen randomly because of little ram, a reboot will recover. I can try to optimize the ram in future verion.

GeekMagicClock, 2024-05-26:
@19201003080114 Hello, please try the latest firmware, if still exists, please reopen issue, thank you.

closed 2024-05-26 by GeekMagicClock