GeekMagicClock/smalltv-ultra #30: Ultra TV firmware update

Azouz77, 2024-04-06:
Seems not possible to update my device with New version v9.0.18 as filé is too Big and i get continuous error 10. Someone has this issue before ??

TrueLoonix, 2024-04-06:
Azouz77 are you using the correct firmware? (Ultra-V9.0.18)
Not sure if this could cause it, but do you have additional pictures saved? Try removing some of them.
Hope it helps, flashing worked flawlessly on my Ultra.

Ashwagada, 2024-04-12:
> Wydaje się, że nie jest możliwa aktualizacja mojego urządzenia do nowej wersji 9.0.18, ponieważ plik jest za duży i pojawia się ciągły błąd 10. Ktoś miał już ten problem?

If you have enough memory, both 9.0.18 and 9.0.19 will update correctly.

8666, 2024-07-31:
I have `Model: Smalltv-Ultra,Version: Ultra-V9.0.16` What can I use as upgrade?

GeekMagicClock, 2024-08-02:
> I have `Model: Smalltv-Ultra,Version: Ultra-V9.0.16` What can I use as upgrade?

V9.0.26 is OK