GeekMagicClock/smalltv-ultra #32: ERROR[10]: Magic byte is not 0xE9

Azouz77, 2024-04-06:
While tryibng to update the device, i get the below error message

ERROR[10]: Magic byte is not 0xE9

I tried many times, but still same error

TrueLoonix, 2024-04-06:
Did you try the same file or did you re-download?
Might be a corrupt file.

Azouz77, 2024-04-06:
I tried with New file but same

GeekMagicClock, 2024-04-07:
Hi, some antivirus software would change the bin file you downloaded. You can try update it with your phone or try shutdown the antivirus software.

Ashwagada, 2024-04-12:
> While tryibng to update the device, i get the below error message
> 
> ERROR[10]: Magic byte is not 0xE9
> 
> I tried many times, but still same error

Always upload the extracted .bin file appropriate to your device model.

fredbear5150, 2024-05-10:
I had the same issue with

ERROR[10]: Magic byte is not 0xE9

until I cloned the repository (in Linux) and uploaded the firmware from there, which had the correct md5sum.

The file is clearly changed slightly if downloaded from the HTTPS link. It has nothing to do with Windows or antivirus because I only use Linux anyway.

GeekMagicClock, 2024-05-10:
> I had the same issue with
>
> ERROR[10]: Magic byte is not 0xE9
>
> until I cloned the repository (in Linux) and uploaded the firmware from there, which had the correct md5sum.
>
> The file is clearly [...]

Which firmware link?  What is the model you have?

fredbear5150, 2024-05-10:
The latest .21 firmware for the Ultra model.

GeekMagicClock, 2024-05-11:
Can you upload a picture about the content displayed on the product ? Most possible reason is wrong firmware for the model.

fredbear5150, 2024-05-11:
Why would me uploading a picture help in any way? I have explained that I solved the problem by cloning the repository and picking up the firmware file from there. I'm done, I'm happy, I have found a workaround and letting everyone else here know that was the solution for me. I hope it works for them also.

No, I won't be uploading a photo because it would prove nothing - but if I take a photo of the device's screen, that's going to show my location and timezone and as someone who works in cyber-security and privacy, I am not willing to do that on a public forum like here.

fredbear5150, 2024-05-11:
PS. It had NOTHING to do with the wrong firmware. My device had the .19 firmware already installed, when I downloaded the .20 and .21 versions, the md5sums did not match - so that was the problem, the files were changed (probably only slightly) when downloaded via the browser, Firefox in my case.

GeekMagicClock, 2024-05-11:
Thanks for sharing.

closed 2024-05-11 by GeekMagicClock

fredbear5150, 2024-05-11:
Do you not understand English? It was the correct firmware, the device is working fine on .21 after I cloned the repository with git. I use Linux and git every day of my life, not that Windows rubbish.

GeekMagicClock, 2024-05-11:
I mean the one you downloaded on your linux is a wrong firmware, content modified, md5sum changed.

fredbear5150, 2024-05-11:
No, it's the right firmware file but the checksum was wrong.