GeekMagicClock/smalltv-ultra #34: ¿Multiple language?

nextvisionmx, 2024-04-10:
Is it possible that it exists in several languages, if the memory is small, something like uploading the language here and being able to load only one language?

GeekMagicClock, 2024-04-10:
Hi, rom/ram both are very limited, possible to add language on the web pages, bu no plan yet.

nextvisionmx, 2024-04-12:
> Hi, rom/ram both are very limited, possible to add language on the web pages, bu no plan yet.

Thanks, but I don't mean the full admin but only what is shown on the screen like days, months, weather information, just what is shown on the screen not the admin.

Ashwagada, 2024-04-12:
> Hi, rom/ram both are very limited, possible to add language on the web pages, bu no plan yet.

If rom/ram are limited, it is best to make separate language versions.
Send us a file with the text and we will translate it into other languages.