GeekMagicClock/smalltv-ultra #35: Can't change location?

Paul12345666, 2024-04-10:
I cannot change the location of the weather app, it lets me check my connection every time I input a new one. Plus my GIF doesn't work in the main page.

GeekMagicClock, 2024-04-10:
Hi, show me your settings on the web?

Paul12345666, 2024-04-10:
[image]

GeekMagicClock, 2024-04-10:
Hi, where did you get it? The chinese version only support cities in China, you may need return and refund.

Paul12345666, 2024-04-10:
I did get this in China, and I'm sure this location that I entered is in China, why it doesn't work? I'm in China right now but am I not able to update the firmware?

GeekMagicClock, 2024-04-11:
> I did get this in China, and I'm sure this location that I entered is in China, why it doesn't work? I'm in China right now but am I not able to update the firmware?

Do not update it.  It's not compatible.

If the device is in China, please go ask for the saler's help, here's for English only.
And that should be a network issue, It's also easy to clarify, try a different network.

closed 2024-04-11 by GeekMagicClock