GeekMagicClock/smalltv-ultra #36: When in 'interval' mode, the 'weather' forecast screen contains no data

Graphnic, 2024-04-10:
When in 'interval' mode, the 'weather forecast' screen contains no data from the feed.  Manually selecting 'weather forecast' manually, the data is loaded. 

Also tried version Ultra-V9.0.19

[image]
[image]
[image]

GeekMagicClock, 2024-04-11:
When the clock is in loop mode, the memory consumption is too large, which may cause some failed requests. You can wait for a while.  In future version, I was considering remove the loop function.

Ashwagada, 2024-04-12:
> When the clock is in loop mode, the memory consumption is too large, which may cause some failed requests. You can wait for a while. In future version, I was considering remove the loop function.

A step towards better optimization. If the loop is unnecessary, it is worth getting rid of it or changing it to another function.

Graphnic, 2024-04-12:
> When the clock is in loop mode, the memory consumption is too large, which may cause some failed requests. You can wait for a while.  In future version, I was considering remove the loop function.

You are right, it does sort itself out after a few minutes. I really like the loop feature, so waiting a few minutes isn't an issue! Thanks for the response:-)

marcelosuzumura, 2024-04-15:
> I really like the loop feature

Me too, it's really useful to cycle automatically between the different views, than just having a fixed one all the time.

VaultBoy88, 2024-04-22:
It's the same here. After upgrading to .20, the forecast is no longer working. Even rolling back to .19 does not help.

Voldenuit23, 2025-05-21:
I very much like the loop function as well.

It might be a good idea to wait for all data to be fetched before switching to the forecast screen.

GeekMagicClock, 2025-05-23:
> I very much like the loop function as well.
> 
> It might be a good idea to wait for all data to be fetched before switching to the forecast screen.

Did you use your own forecast API already?