GeekMagicClock/smalltv-ultra #38: Suggestions and errors v2

Ashwagada, 2024-04-12:
Hello,

You haven't responded to the previous ticket yet, but here are my suggestions and errors v2.

I am counting on feedback on each of the suggestions and errors mentioned.

I would probably have even more suggestions and bugs, but I don't know if resources will allow it.

Version: Ultra-V9.0.19

Suggestions:

1. Security:
1.1. To avoid unauthorized access to the management page, I recommend entering an account system or just a password.
1.2. I don't know if it's possible, but you could consider introducing an SSL certificate (https).
1.3. On the website, in the "Network" tab, the password should be replaced with * with the option of revealing (as is the case with most websites).

2. In the "Network" tab there should be a bar with the network coverage strength, just like in the case of search.
(Re-checking the strength every page refresh.)
(It would also be better to change the displayed strength to %.)
[image]

3. In the "Settings" tab, the "Brightness" function is set using a slider. It would be useful to have an additional option to manually enter (0-255 or 1-100%).

4. Information about the new software version would be useful on the management page.

5. In the "Settings" tab, after clicking "RESET" or "REBOOT", a window should appear asking if you really want to do this.
(In fact, something like this would be useful everywhere or "Save" to avoid accidental changes.)

6. Delete unnecessary images/gifs taking up disk space.
[image]

7. Try to optimize RAM and ROM to ensure greater stability and responsiveness.

8. Maybe it's worth switching from "OpenWeatherMap" to something else?

9. If you can't introduce multilingualism, maybe it's worth making a separate version of the software for each country?
(I can translate it into Polish because I and others care about it very much.)

10. The management page works very quickly, but its appearance could be improved to be more modern and up-to-date.

11. Think about better placement and sizes so that there are no situations in which something is not displayed.
(As with the name of a country next to a city.)

Errors:

1. Sometimes, after entering the management page, the message "Device connection lost, check connection and reload please..." appears, even though the connection is correct.
(After refreshing the management page, everything works.)
[image]

2. Sometimes there is a problem with updating the date (on its own or after a reset)
The default setting is "1/1/1970".
(After some time, the correct one is displayed.)

3. Longer city names do not display the country name next to them.

PS. You can change to 2024 ;)
[image]

marcelosuzumura, 2024-04-15:
> In the "Settings" tab, the "Brightness" function is set using a slider. It would be useful to have an additional option to manually enter (0-255 or 1-100%).

I would be very interested in this feature too, setting it via mouse is very innacurate.

GeekMagicClock, 2024-04-15:
> > In the "Settings" tab, the "Brightness" function is set using a slider. It would be useful to have an additional option to manually enter (0-255 or 1-100%).
> 
> I would be very interested in this feature too, setting it via mouse is very innacurate.

Got it! Thank you all.