GeekMagicClock/smalltv-ultra #39: API key

Ashwagada, 2024-04-12:
Hi,

I use my own API key (from what I remember, the free version has 1000 queries)

Can I set the minimum refresh rate to every 1.5-2 minutes?

If I understand correctly, there are 1440 minutes in a day, so refreshing every 2 minutes should use 720 queries?

Alternatively, what do you recommend?

I currently have it set to every 5 minutes, but I want the weather to update as quickly as possible.

GeekMagicClock, 2024-04-12:
Yes, you could, please update to v9.0.19 version, it's limited to at highest 1 minute.

Ashwagada, 2024-04-12:
I am using the latest version of the software.

Enough queries to refresh every minute?
Is it better to set it to every 2 minutes?

GeekMagicClock, 2024-04-13:
> I am using the latest version of the software.
> 
> Enough queries to refresh every minute? Is it better to set it to every 2 minutes?

Yes, 2 minutes is OK.

closed 2024-04-20 by GeekMagicClock