GeekMagicClock/smalltv-ultra #40: Source Code

fbnlrz, 2024-04-12:
Hi :) First of all i love this cute little Device!

But could you Pleeeease provide the Sourcecode? 
A lot of people would love to integrate it into Home Assistant or do cool stuff with it.
Thank you!!

Ashwagada, 2024-04-12:
Good idea.

vjhoming, 2024-04-13:
Yes please !

MaximNedavniy, 2024-04-13:
Source code is not only customization, but also security (it is possible to see what a device connected to the network is actually doing).

BLKFSH, 2024-04-23:
Agreed! I'd love adding some custom features and while I could just wipe it and make my own firmware, it would be cool to have the ability to keep the current one :)

JoeB2, 2024-04-25:
I am confident the the +1hr time bug and the the DST flag inop bug could be quickly found and corrected by many people in this community.  i.e. If only we had source code access.  Please consider providing us with source code access.

taborrobak, 2024-05-02:
+1
a) Can't keep this on my network until I know what's going on under the hood
b) Might as well make it opensource so the community can develop it

pakair, 2024-05-19:
source code would be nice ... 
I have an esp32 and a 3.2 inch SPI display (ILI9341) and I'd like to make the code work on a bigger display.

Luanvbn, 2024-07-10:
Please make the source code available, I would also like to customize it with my Alexa and other things.

ferenczy, 2024-08-12:
I'll just add that by opening the source code, the device manufacturer would also benefit because they will get a **"community developer" working 24/7 on new features, improvements, and bug fixes for free, making the device more attractive to potential buyers**. If it would attract a reasonably sized community, it may turn the device into a multi-purpose one, that may be used for all kinds of purposes. In contrast to the current state, which is cool, don't take me wrong, but it's just a simple clock with weather information, so nothing particularly interesting. The value is in the hardware, so let's take it to another level by inflating the value of the software component. And as a consequence, boost the device sales! What do you think, @GeekMagicClock, please?

ferenczy, 2024-08-13:
Imagine creating a simple app store, where you can install various apps/plugins with a mouse click. Just by installing an additional app/plugin, the device would turn into a little Home Assistant dashboard. Or display cryptocurrency/fiat exchange rates. Or show a subscriber counter from social platforms. Basically, similar to how the LaMetric device works, for example.

rob040, 2026-02-15:
+1
This is indeed a great device for adding features by open source community , although the PRO (ESP32) version would be better suited for adding 'apps' etc. It has more RAM and can be had with more flash. Also, the core development of ESP8266 is stopped for many years.

Similar open-source requests are in issue #56, #90, #157, #126, #92, and on other issues.