GeekMagicClock/smalltv-ultra #44: Pictures shown much brighter than other views

marcelosuzumura, 2024-04-15:
v9.0.19 (and previous one too)

I'm using auto switch themes and included photo album in rotation. Also configured night mode with a very dim brightness.

When it's night, all themes are dim, but when showing the picture album, it seems to pick the daylight brightness and it's shown way brighter.

Configs:
[image]
[image]

Thanks!

GeekMagicClock, 2024-04-15:
Thanks for report, will be fixed in next update.

GeekMagicClock, 2024-04-20:
@marcelosuzumura  Please update to V9.0.20, thanks.

marcelosuzumura, 2024-04-21:
@GeekMagicClock nice, thanks! the image seems as bright as other views now during night time.

Now there's just an issue that I have 4 images in the rotation, but when in night time, it only shows the last image of the set. The time allocated for the first 3 images shows a black screen (screen off?) instead. During the day they show normally as expected.

GeekMagicClock, 2024-05-26:
> @GeekMagicClock nice, thanks! the image seems as bright as other views now during night time.
>
> Now there's just an issue that I have 4 images in the rotation, but when in night time, it only shows [...]

Hello, sorry this reply maybe too late, what are the format of these for images? Are the first three gif files?

marcelosuzumura, 2024-05-26:
@GeekMagicClock hey, no problem. All images are static, no gifs used.

closed 2026-02-01 by GeekMagicClock