GeekMagicClock/smalltv-ultra #48: Operate in offline mode

dansikk, 2024-04-21:
I would like to upload just a few pictures and display them in offline mode. How is that possible? I don't want a connection to the Internet

GeekMagicClock, 2024-04-22:
Well, I can understand your request, I'm trying to offer an option.

Pucur, 2024-04-22:
With a good router like MikroTik you can specify the outbound connections for the devices, also you can block it. Or set up a dns server and you can block all the outbound requests

dansikk, 2024-04-22:
Thanks for the tip, but unfortunately I don't have a WiFi connection at work. I want to display 1-2 pictures of my cat on the screen...
Is there perhaps another option?

Pucur, 2024-04-23:
Share your phone wifi then and you can upload the pictures what you wanted. Then its not necessary I think to have Wifi access for the device, you can set it up to rotate the images only.

dansikk, 2024-05-19:
After every restart the display searches for internet. Would an update to the firmware be possible in order to operate the display completely without the Internet?

Ashwagada, 2024-05-19:
> After every restart the display searches for internet. Would an update to the firmware be possible in order to operate the display completely without the Internet?

```
V9.0.22
5) Feature：Photo album theme could work without internet connected. [AP "GIFTV" will start for 1 minute after power on, if there's no clients connected, the "GIFTV" will be closed.]
```

cconde, 2024-05-21:
Potentially it could have an Internet timeout option and then go into AP mode *while* performing it's usual function, replacing those screens that require internet connection for a "No internet" message.

GeekMagicClock, 2024-05-26:
> Potentially it could have an Internet timeout option and then go into AP mode _while_ performing it's usual function, replacing those screens that require internet connection for a "No internet" message.

Only the photo album could work without internet, added in newest firmware, for theme related to weather and clock, it is internet needed.

closed 2024-05-26 by GeekMagicClock