GeekMagicClock/smalltv-ultra #49: changing the weather server

Dmitry-systemd, 2024-04-22:
it would be convenient to replace the weather server because the current one displays approximate data

GeekMagicClock, 2024-04-23:
Any recommended server ?

Dmitry-systemd, 2024-04-23:
met.no

https://github.com/metno/weatherapi-docs

Ashwagada, 2024-04-23:
Better

weather.com
https://www.wunderground.com/weather/api

AccuWeather
https://developer.accuweather.com/

GeekMagicClock, 2024-04-24:
> Better
> 
> weather.com https://www.wunderground.com/weather/api
> 
> AccuWeather https://developer.accuweather.com/

Seems these two sites  provide very limit free access api.

JoeB2, 2024-04-25:
I use https://www.wunderground.com/weather/api for the western half of the US for small airplane weather.  Weather Underground  has proven itself, from my usage, to be the most consistently accurate.  **One vote for** replacing the devices weather source with ### **https://www.wunderground.com/weather/api**
Thank you,
joe

rwkyyy, 2024-04-26:
Foreca is the best, it shows the nearest weather station to your location, in big cities almost all show a "median"

GeekMagicClock, 2024-06-18:
> @GeekMagicClock openweathermap is, unfortunately, changing their API and licensing, making the 'free tier' only possible when committing to pay for additional calls on top of the free allotment. Are there plans to add other providers?

Any suggestions? Seems like the options are very limited.

Habbie, 2024-06-19:
Besides thinking about other services, how about thinking about "it is likely somebody will launch another service that is compatible with openweathermap - some geeks might even run one in their house". The only thing necessary to support that would be to enter a custom URL for an openweathermap-compatible service.

rvdm, 2024-06-19:
> Besides thinking about other services, how about thinking about "it is likely somebody will launch another service that is compatible with openweathermap - some geeks might even run one in their [...]

I think this is the way. It would be relatively easy to fix in the code, and it gives us long term options. I like the idea of adding other weather services, but I'd quickly end up suggesting ones local to me - or other (rate or cost) limited options. 

@GeekMagicClock is that easily doable? I think there's multiple people that wouldn't mind typing it up should that be possible at some point :)

GeekMagicClock, 2024-06-20:
Yes,  if the api is compatible with openweathermap.org, then it would be easy to add.

rvdm, 2024-07-04:
> Yes, if the api is compatible with openweathermap.org, then it would be easy to add.

Nice! Is there any help we can give to add the 'custom openweathermap API URL' option?

GeekMagicClock, 2024-07-04:
> > Yes, if the api is compatible with openweathermap.org, then it would be easy to add.
> 
> Nice! Is there any help we can give to add the 'custom openweathermap API URL' option?

Should be http request and response like this : {"coord":{"lon":-0.1257,"lat":51.5085},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"base":"stations","main":{"temp":19.04,"feels_like":18.17,"temp_min":17.94,"temp_max":20.29,"pressure":1009,"humidity":45,"sea_level":1009,"grnd_level":1005},"visibility":10000,"wind":{"speed":8.23,"deg":260},"clouds":{"all":5},"dt":1720095506,"sys":{"type":2,"id":2075535,"country":"GB","sunrise":1720064999,"sunset":1720124376},"timezone":3600,"id":2643743,"name":"London","cod":200}