GeekMagicClock/smalltv-ultra #50: Power Off Power On - Unit not updating

JoeB2, 2024-04-25:
0) weather and time and display settings all work as configured.  Unit works as advertised.
1)  I unplugged the USB power the unit shutoff immediately
2) Plugged in USB power & unit booted up
3) unit shows correct IP address & that IP does have internet access (tested)
4) unit shows 00:00 00 as startup time.  Seconds and minutes count up from 1/1/1970 00:00 00.
   verified NTP configuration : correct as per firmware
5) wind, pressure, rain all show 0.  temp shows 32 f
6) Connect to unit's configuration web pages using the IP works as designed
7) Changes to any configuration via the web pages  works as designed.  i.e. IP connects to config web pages and settings work
8) Reboot / Reset : the unit remains in its non-updated state.  i.e. 1/1/1970 00:00 00 ; No Weather update

Unit is inop

Ashwagada, 2024-04-25:
I recently wrote about this error here: https://github.com/GeekMagicClock/smalltv-ultra/issues/38

2. Sometimes there is a problem with updating the date (on its own or after a reset)
The default setting is "1/1/1970".
(After some time, the correct one is displayed.)

JoeB2, 2024-04-26:
Hi Ashwagada,
Its been several days and still no change.  Date is 1970 and weather shows 32f 0 rain (despite all the rain over the past two days).  The unit remains inop.  I would like to clear the SPIFs/LittleFS - please LMK which file system is in use.  BLW: I am sure this is something I can fix, if only I had access to the source.  IDF/Platformio/Arduino IDE ?

Thank you for your response,
joe

GeekMagicClock, 2024-04-27:
Hi, if the date stays at 1970, the most possible reason is there's network limit, you can try with another network.