GeekMagicClock/smalltv-ultra #55: Delay WiFi Connection

Ashwagada, 2024-05-06:
Hi,

the website defaults to 30 seconds, but it's actually 1 second.

Only after changing it to e.g. 60 seconds and returning to 30 does it become 30 seconds.

By the way, how do I set the delay when I have the router automatically restarts every day?

Setting it to 150 seconds doesn't satisfy me because when I restart smalltv I have to wait 5 minutes for the connection.

Maybe it's worth adding a delay when restarting smalltv and when the network goes out? Or after losing Wi-Fi, it will try to connect, e.g. every few seconds/minutes

GeekMagicClock, 2024-05-06:
Hi, the delay works if the smalltv automatically restarts every day with the router.
But if the smalltv does not restart, the delay will not work.

Yes, it's worth to add a re-connect strategy inside if connection lost, I'll add it to my todo list to the next updates, thanks for letting me know your needs.

Ashwagada, 2024-05-06:
Thanks, but take a look at the error I wrote above.

GeekMagicClock, 2024-05-06:
OK, that should be a web code error, default should be 0, will fix in next update, thank you!

Ashwagada, 2024-05-06:
> Hi, the delay works if the smalltv automatically restarts every day with the router. But if the smalltv does not restart, the delay will not work.

It would be nice if you added an option to automatically restart at a time of our choosing.