GeekMagicClock/smalltv-ultra #56: Advocating for open sourcing firmware for enhanced customization

teumaas, 2024-05-06:
Hi there!

Firstly, I absolutely love your product! However, I believe there's untapped potential in allowing users to customize the firmware. Open-sourcing the firmware would not only empower users to tailor the product to their specific needs but also foster a vibrant community of developers collaborating to enhance its capabilities. Personally, I'm eager to add some unique features that would elevate my user experience and potentially benefit others as well. Let's unlock the full potential of this fantastic product by embracing open-source principles!

Atari361, 2024-05-07:
+1

I'd personally like to migrate to another "weather" service, a more local one that has more accurate results for my location.
Maybe also implement small improvements overall, and not wait for an official release.

cconde, 2024-05-15:
+1 This device has huge potential if we can add/modify features

jo-m, 2024-05-16:
I have created a sample project showing how to use the display: https://github.com/jo-m/smalltv-pro-esp-idf

[This one](https://github.com/jo-m/netshlix/tree/main/main) also spins up the WiFi (although with hardcoded credentials)

patrick-othmer, 2026-01-07:
+1 Opensource would be great! Think of all the crypto bros.