GeekMagicClock/smalltv-ultra #58: Not updating weather/time info after connecting

umbra2707, 2024-05-10:
Hello! 

I got a smalltv Ultra. I got connection and location set up but it doesn't update the info on the display.

The device connects to my wifi as it should and I can reach it through a browser. I've used the code for my city an the example code but nothing happens. I even got an API Key from OpenWeather but it's the same. 

[image]

Am I doing something wrong? I'm currently using the latest version (Model: Smalltv-Ultra,Version: Ultra-V9.0.21)

GeekMagicClock, 2024-05-11:
Yes, something is wrong with your settings, show me your settings on the web please, the city code area.

umbra2707, 2024-05-11:
I could get it working but only on my Android phone hotspot. When I change to my WiFi router it stops working again.

GeekMagicClock, 2024-05-26:
> I could get it working but only on my Android phone hotspot. When I change to my WiFi router it stops working again.

This should be a network issue, is there any blacklist or whitelist on your router?