GeekMagicClock/smalltv-ultra #59: IP address after reconnecting to the network

Ashwagada, 2024-05-11:
Hi,

the last update, i.e. V9.0.21, helped me a lot when restarting the router every day, but after reconnecting the device it shows a strange IP address, not the one it was before.

Additionally, it would be nice to be able to set your own reconnect time, rather than a fixed 5 minutes.

GeekMagicClock, 2024-05-11:
Hi, the IP is assigned by the router, you can set a static IP for the clock.
I just want to keep it simple, if the reconnect time is too short, there will be problems, there may be strange behavior.

Ashwagada, 2024-05-11:
So why is the IP address different when I reconnect to the network, but when I manually restart the device it remains constant, which is what it should be?

GeekMagicClock, 2024-05-11:
You may have seen the "IP unset" string, that means the wifi connection is still lost.

Ashwagada, 2024-05-11:
There was "IP unset", then a different IP address was shown, and after a manual restart everything returned to normal.

GeekMagicClock, 2024-05-11:
> There was "IP unset", then a different IP address was shown, and after a manual restart everything returned to normal.

Hi, this should not happen, it will display "IP unset" or real IP, but will not display "IP unset" and then a different IP address.

Ashwagada, 2024-05-11:
And yet it is so, what can we do about it?

GeekMagicClock, 2024-05-12:
> And yet it is so, what can we do about it?

You can check the connection after 5 minutes, it should be recovered, and display the correct IP address.

Ashwagada, 2024-05-12:
Another restart and the same situation.

I noticed on the router's website that there is no smalltv among the connected devices, so as I said, reconnecting after 5 minutes is faulty.

GeekMagicClock, 2024-05-12:
Hi, how that happened? You turned off the router, and the IP  displayed like "IP unset",and then you turned on the router,  the IP should not recovered after 5 minutes, is that so?

Ashwagada, 2024-05-12:
I turned off the router and "IP unset" is displayed, then I turned on the router and after 5 minutes a different IP address is shown and smalltv is not connected to the network.

GeekMagicClock, 2024-05-12:
> I turned off the router and "IP unset" is displayed, then I turned on the router and after 5 minutes a different IP address is shown and smalltv is not connected to the network.

Hi, a "different IP address", what it is like?

Ashwagada, 2024-05-12:
> > I turned off the router and "IP unset" is displayed, then I turned on the router and after 5 minutes a different IP address is shown and smalltv is not connected to the network.
> 
> Hi, a "different IP address", what it is like?

It shows 169.254.17.67 but is not connected to the internet.

GeekMagicClock, 2024-05-12:
Thanks for the info, seems the device automatically assigns an IP address to itself when it cannot obtain an IP address from a DHCP (Dynamic Host Configuration Protocol) server. This usually occurs when there is a network connection problem, such as an unreachable DHCP server or a network misconfiguration.

What kind of router do you have? Can you help show me your wireless setting options on your router?

Ashwagada, 2024-05-12:
Funbox 6 (DHCP) -> Archer AX55 (AP 2.4GHz - DHCP off).

What settings are you interested in?

GeekMagicClock, 2024-05-12:
> Funbox 6 (DHCP) -> Archer AX55 (AP 2.4GHz - DHCP off).
> 
> What settings are you interested in?

Like wireless mode b/g/n, working channel etc.

Ashwagada, 2024-05-12:
On:
- OFDMA,
- WMM,
- Airtime Fairness.

Security: WPA3-Personal+WPA2-PSK[AES]
Channel width: 20/40MHz
Channel: Automatic
Mode: Mixed 802.11b/g/n/ax

GeekMagicClock, 2024-05-12:
OK, can you help try a test firmware ?
[Uploading firmware-test-v9.0.22.zip…]()

Ashwagada, 2024-05-12:
Unable to download, refreshes the page.

GeekMagicClock, 2024-05-12:
> Unable to download, refreshes the page.

Oh, sorry, please try again.

Ashwagada, 2024-05-12:
It still shows 169.254.17.67.
It clearly cannot connect to my network after 5 minutes.
Only disconnecting and reconnecting the power helps.

GeekMagicClock, 2024-05-13:
OK, can you help confirm which time the connect successed after the power on?  The progress bar will try 3 times, is it the first time or second or the third time ?

Ashwagada, 2024-05-13:
https://github.com/GeekMagicClock/smalltv-ultra/assets/165925449/0d165103-3a53-4ed8-a315-eabd08b02835