GeekMagicClock/smalltv-ultra #63: Update ERROR[10]: Magic byte is not 0xE9

eugurlu, 2024-05-18:
I have geekmagic weather clock Smalltv-Ultra V9.0.19
I want to upload new firmware. (V9.0.22)
I download from https://github.com/GeekMagicClock but I get 
Update error: ERROR[10]: Magic byte is not 0xE9 error each time I try to update.
I tried different firmware versions but always same error. 

Any ideas how to update this one?
[image]

Ashwagada, 2024-05-18:
Hi,

follow this link - the latest version (V9.0.22) will be automatically downloaded.
https://github.com/GeekMagicClock/smalltv-ultra/raw/main/Ultra-V9.0.22/FW-Smalltv-Ultra-V9.0.22.bin
Then try to update your device.

If this fails, try downloading and updating from your mobile device.

eugurlu, 2024-05-18:
downloading directy from this link solved it. Thank you very much for your help!

Ashwagada, 2024-05-18:
Can you tell me how you downloaded it before? Was it just a file or a ZIP/RAR? What did you use to unpack?

eugurlu, 2024-05-18:
https://github.com/GeekMagicClock/smalltv-ultra/blob/main/Ultra-V9.0.22/FW-Smalltv-Ultra-V9.0.22.bin ->I downloaded from here which was on the page: https://github.com/GeekMagicClock/smalltv-ultra/tree/main/Ultra-V9.0.22

Ashwagada, 2024-05-18:
It appears to be a repository issue, as downloading the .bin file from the link I sent you works fine.

It is important to provide as much information as possible so that the author of the project can repeat it and possibly find a solution.

To be sure, make a video showing exactly how you download the software, because the link doesn't explain much.