GeekMagicClock/smalltv-ultra #66: Gif not updated

DerSarco, 2024-05-23:
Hello! i was updating my SmalTv Ultra to the latest firmware, after that i wanted to change the gif that the screen display at the weather view. I cleaned the files because i didn't have enough ROM to store a new gif, but after uploading the gif, the screen doesn't show it anymore. I used the gifs at 80x80 that you have in a repository. i miss my astronaut :(

[image]

ticsi, 2024-05-23:
I had a similar issue with the latest firmware. I'm looping between a full screen gif and the weather display with the small gif. After around 20 minutes or so, both gifs stop showing up. If I reboot, it works again.

(Also, if I set this automatic switching of the screens, the full-screen gif only plays once, then freezes. I'd like it to loop until the screens change.)

GeekMagicClock, 2024-05-23:
@DerSarco  Have you upload a new gif to the weather options, not under the photo album option.

@ticsi  V9.0.22 right ?

ticsi, 2024-05-23:
> @ticsi V9.0.22 right ?

Yes, that's right, I've updated today.

[image]

DerSarco, 2024-05-23:
> @DerSarco Have you upload a new gif to the weather options, not under the photo album option.

Yes, i updated to the last firmware also. But the gif it doesn't display. I tried uploading to the photos section, just to test, but is the same, nothing is shown.

GeekMagicClock, 2024-05-24:
> > @DerSarco Have you upload a new gif to the weather options, not under the photo album option.
> 
> Yes, i updated to the last firmware also. But the gif it doesn't display. I tried uploading to the photos section, just to test, but is the same, nothing is shown.

Show me your file list please.

DerSarco, 2024-05-24:
> > > @DerSarco Have you upload a new gif to the weather options, not under the photo album option.
> > 
> > 
> > Yes, i updated to the last firmware also. But the gif it doesn't display. I tried uploading to the photos section, just to test, but is the same, nothing is shown.
> 
> Show me your file list please.

It's weird, because i upload the image, the ROM is used but i don't have a list of images.

[image]

[image]

GeekMagicClock, 2024-05-26:
Hello, can you try the newest firmware V9.0.23, and also could you please provide the file with the same name you uploaded, I can try on my side.

DerSarco, 2024-05-26:
> @ticsi @DerSarco Hello, can you try the newest firmware V9.0.23, and also @DerSarco could you please provide the file with the same name you uploaded, I can try on my side.

Ok, i'll try. The files that i used are the next ones ->

[image]
[image]
[image]

DerSarco, 2024-05-26:
Ok, i updated to the .23 version, i cleared the files from pictures and weather, i uploaded the spaceman again but it doesn't work. i'm crying inside 😢

GeekMagicClock, 2024-05-27:
@DerSarco 
[image]
OK, have you click the "SET" button?

DerSarco, 2024-05-27:
Ok, i needed to refresh the page to see the set button. Awesome, now that's working! thanks!

closed 2024-05-27 by GeekMagicClock