GeekMagicClock/smalltv-ultra #67: Forecast screen is broken in 9.0.22

ticsi, 2024-05-25:
Forecast screen stops updating after a few hours. (You can tell because the clock is getting stuck in the top row.)

If I set the forecast screen and reboot, it never loads anything, like in the picture.

[image]

GeekMagicClock, 2024-05-26:
Thank you for the report, please update to v9.0.23.

ticsi, 2024-05-26:
Thanks! I've updated. The weather forecast is now fixed after reboot, but now the time in the top row does not update at all.

GeekMagicClock, 2024-05-26:
> Thanks! I've updated. The weather forecast is now fixed after reboot, but now the time in the top row does not update at all.

Not update? Always the same time? Did you enabled the theme loop or not?

ticsi, 2024-05-26:
I didn't enable the theme loop. I think if I did, it would work, because when I switch to the forecast screen it shows the current time, but then stays the same, the minutes don't increase. Other themes seem to work fine.

GeekMagicClock, 2024-05-26:
> I didn't enable the theme loop. I think if I did, it would work, because when I switch to the forecast screen it shows the current time, but then stays the same, the minutes don't increase. Other themes seem to work fine.

Thanks, it's a bug, here is a quick fix. 
[FW-Smalltv-Ultra-V9.0.24.zip](https://github.com/GeekMagicClock/smalltv-ultra/files/15447448/FW-Smalltv-Ultra-V9.0.24.zip)

ticsi, 2024-05-26:
That's amazing support, thank you.

So now what happens is, if I boot to weather forecast, it works perfectly. However, if I switch to another screen than back to weather forecast, it's completely blank at first, then after a minute, the clock appears, but not the weather.

[image]

I appreciate your dedication to this small gadget, take your time with the fix, I'm not impatient. :)

GeekMagicClock, 2024-05-26:
> That's amazing support, thank you.
> 
> So now what happens is, if I boot to weather forecast, it works perfectly. However, if I switch to another screen than back to weather forecast, it's completely blank at first, then after a minute, the clock appears, but not the weather.
> 
> ![P_20240526_152020](https://private-user-images.githubusercontent.com/32036814/333882408-51ffdc7b-6929-4320-8b3f-32b53c3be333.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3MzA3NDYsIm5iZiI6MTcxNjczMDQ0NiwicGF0aCI6Ii8zMjAzNjgxNC8zMzM4ODI0MDgtNTFmZmRjN2ItNjkyOS00MzIwLThiM2YtMzJiNTNjM2JlMzMzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI2VDEzMzQwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg1ZmQ0MTQ2YTMwY2FlOThhMTgwYjJmMTRjNGYzYzA4ZDYyZDQyYTVkOWUxZWIzNWMyYTc5MjUxMTEyZmZmMWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DsujqHAyZzGCbqOhdyKomCNwFyGIpltzaHdQQc_hL0M)
> 
> I appreciate your dedication to this small gadget, take your time with the fix, I'm not impatient. :)

Got it, thank you,  I'll send you another one tomorrow, cause it's late here.

GeekMagicClock, 2024-05-27:
[FW-Smalltv-Ultra-V9.0.24.zip](https://github.com/GeekMagicClock/smalltv-ultra/files/15449737/FW-Smalltv-Ultra-V9.0.24.zip)
Hello, I think this time should be fixed :)

ticsi, 2024-05-27:
> [FW-Smalltv-Ultra-V9.0.24.zip](https://github.com/GeekMagicClock/smalltv-ultra/files/15449737/FW-Smalltv-Ultra-V9.0.24.zip) Hello, I think this time should be fixed :)

Fixed! 😄 Thanks so much. Keep up the great work!

closed 2024-05-27 by ticsi