GeekMagicClock/smalltv-ultra #70: No WiFi AP after the first power on

ferenczy, 2024-05-29:
I had an issue with a new device, even though it seemed to be working properly after I powered it on for the first time - it noticed me on the display, that it failed to connect to a WiFi network so it has switched to an AP mode and I should connect to the WiFi network `GIFTV`. But it actually didn't create any AP, I have checked it using multiple devices. I have tried to restart it multiple times, but it was still the same. I started to have a suspicion that it was simply broken and almost gave up.

But then I tried to connect it to a different power source (using the same USB cable) and it suddenly successfully created the AP `GIFTV`! It's weird, but the change of a power source seems to have fixed the issue. Maybe the first power source didn't supply enough power for the radio part of the product.

Anyway, it would be good to mention this issue somewhere (web, manual, etc.) because according to comments on AliExpress, several people had the same issue that it didn't create the AP after the first power on.

GeekMagicClock, 2024-05-29:
Yes, that's correct, we have encountered this problem and find the solution as what you said, power related issue, I'll try to add it to the manual's FAQ part, thank you so much.

ferenczy, 2024-06-01:
Thank you @GeekMagicClock! I think that It may help prevent devices from ending up in a trash bin. I had discovered it just by chance, right before I was almost ready to give up. And it would be a pity to trash it because it's otherwise a very neat device. ❤️

Also, it's not an issue just with the AP mode, but with the WiFi altogether. With some power sources, the device seems to be working properly (it starts up, and displays information on the screen), but it neither creates an AP nor connects to a WiFi network when it's already set up.

ferenczy, 2024-06-04:
@GeekMagicClock BTW do you know what exact property of a power source is causing the issue? Is it the voltage (e.g. the WiFi stops working when the input voltage drops under 5 V)? I doubt the current may cause problems since the device draws very little (typically around 100 mA, up to 200 mA during the startup), so de facto any power source should be sufficient, even a computer USB port.

Also, I have noticed that even when the device starts properly and successfully connects to a WiFi network, it may happen that it will lose the connection later (it loses the IP address so maybe the whole radio part comes down).

GeekMagicClock, 2024-06-05:
Hi, I consulted the module supplier and their reply was that the voltage was too low and the power supply was insufficient, will cause uncertain problems, like WiFi hostpot lost, connection lost.