GeekMagicClock/smalltv-ultra #71: Unable to connect to the device

Caramassa76, 2024-05-29:
Hi,
I’ve purchased two devices. Where i couldn’t connect to the first one (found the Giftv wifi signal on my Iphone but it keeps trying to connect without success) the second one worked immediately !
I’ve tried to switch to another power source, cable but nothing worked

GeekMagicClock, 2024-05-30:
> Hi, I’ve purchased two devices. Where i couldn’t connect to the first one (found the Giftv wifi signal on my Iphone but it keeps trying to connect without success) the second one worked immediately ! I’ve tried to switch to another power source, cable but nothing worked

> Hi, I’ve purchased two devices. Where i couldn’t connect to the first one (found the Giftv wifi signal on my Iphone but it keeps trying to connect without success) the second one worked immediately ! I’ve tried to switch to another power source, cable but nothing worked

Hi, in most cases, replace with another power adapter can help, or try with android phone or a computer.

caramassa, 2024-05-30:
My computer doesn't even find the wifi network GIFTV! I've plugged it directly to the PC and it doesn't show up neither
I've already tried with another power source. unfortunately i d'on't have an Android device

GeekMagicClock, 2024-05-30:
#70   It's quite similar to this problem, related with the power source on some boards, or you may return and get a new one.