GeekMagicClock/smalltv-ultra #73: Current *Time Number Color* config should be read from `timecolor2.json`

guillermocalvo, 2024-06-04:
I'm using the latest version of the firmware:

- Model: `SmallTV-Ultra`
- Version: `Ultra-V9.0.24`
 
I noticed the frontend side tries to retrieve the current *Time Number Color* config from `timecolor.json`, but the backend side stores it as `timecolor2.json`.

This problem can be easily fixed like this:

>
> ### File `time.html`
>
> ```diff
> <HTML>
>   <head>
>     <meta charset="utf-8">
>     <meta name="viewport" content="width=device-width, initial-scale=0.8, minimal-ui">
>     <meta name="author" content="GEEKMAGIC">
>     <meta name="theme-color" content="#36393E" />
>     <link rel="stylesheet" href="css/style.css">
>   </head>
> -  <body onload = "getData('/hour12.json');getData('/ntp.json');getData('/day.json');getData('/timecolor.json');getData('/font.json');getData('/timebrt.json');getData('/colon.json');getData('/dst.json');getData('/daytimer.json');getNav();">
> +  <body onload = "getData('/hour12.json');getData('/ntp.json');getData('/day.json');getData('/timecolor2.json');getData('/font.json');getData('/timebrt.json');getData('/colon.json');getData('/dst.json');getData('/daytimer.json');getNav();">
>     <nav id="nav" style="width: 80%;margin: 0 auto;font-size: 25px;text-align: center;">
>     </nav>		
>     ...
> ```

By the way, it would be great if we could use pull requests to propose bugfixes. Other users have already requested [access to the source code](https://github.com/GeekMagicClock/smalltv-ultra/issues/40) and I would happily contribute to the project if it was available on GitHub.

teumaas, 2024-06-06:
I've already created an issue for this, but unfortunately, the owner of this project ignores everything related to open sourcing this product, probably due to his own financial interests. Very sad...

GeekMagicClock, 2024-06-07:
Open source project please refer to GeekMagic S3 [here](https://github.com/GeekMagicClock/GeekMagic-S3).