GeekMagicClock/smalltv-ultra #74: Ask to add bitcoin price ticker.

dhkkim, 2024-06-04:
Please add support for bitcoin price ticker. Thanks.

GeekMagicClock, 2024-06-29:
That is part of the smalltv-pro model, ultra will not be added for the time being.

Jungdol, 2024-07-03:
I hope Bitcoin charts are implemented quickly! 

It looked similar to the pro, so I thought it would work and bought two 😂

cryptoplas, 2024-09-15:
> I hope Bitcoin charts are implemented quickly!
> 
> It looked similar to the pro, so I bought two thinking it was a supported device 😂

Same happened to me. Please can  you implement Bitcoin price ticker? Thanks in advance.

GeekMagicClock, 2024-09-17:
Again, sorry, no plans.

kendoskany, 2024-12-03:
I bought it as well without knowing of differ versions, now it seems like a scam. You should change your plans about ultra version.