GeekMagicClock/smalltv-ultra #77: Request: Sunrise/Sunset screen

Duberry, 2024-06-11:
openweather JSON includes the sunrise and sunset

```
 "sys": {
    "type": 2,
    "id": 2075663,
    "country": "IT",
    **"sunrise": 1661834187,
    "sunset": 1661882248**
  },
```

Request to add a simple sunrise and sunset time screen?

GeekMagicClock, 2024-06-12:
Hi, thank you, how to convert UNIX sunrise/sunset time to your local time hours and minutes?

Duberry, 2024-06-15:
> Hi, thank you, how to convert UNIX sunrise/sunset time to your local time hours and minutes?

void convertUnixToLocalTime(time_t unix_timestamp) {
  struct tm timeinfo;
  localtime_r(&unix_timestamp, &timeinfo);

  // Print the converted local time
  Serial.printf("Converted local time: %04d-%02d-%02d %02d:%02d:%02d\n",
                timeinfo.tm_year + 1900,
                timeinfo.tm_mon + 1,
                timeinfo.tm_mday,
                timeinfo.tm_hour,
                timeinfo.tm_min,
                timeinfo.tm_sec);
}