GeekMagicClock/smalltv-ultra #91: [Request] Not to go back to Reset State (W

poucotm, 2024-07-15:
Actually, WIFI setting is impossible at the office, because user WIFI is blocked for security reason.
But I want 'GeekMagic' to keep the settings at the office after power on. As you know, when it is power on, it searches WIFI connection first, and if it fails, it goes back to WIFI setting mode. I don't need it to update time, and weather information at the office after I get it set at home. How to

GeekMagicClock, 2024-07-16:
Hi, what do you want to display when internet is not available?

poucotm, 2024-07-16:
[image]

As long as it displays the time correctly, it's fine. I want it to keep the weather/temperature information which it gets last time. 😁

GeekMagicClock, 2024-07-19:
Hello, if in that way, weather is not accurate, and time is wrong too, which means nothing, the weather and time both are internet needed.

poucotm, 2024-07-20:
Hi, I know that. I use it for its beauty rather than for the weather information. 
if not, I can only see the wifi setting mode screen at the office.