GeekMagicClock/smalltv-ultra #94: Problem in upload gifs

RTormente, 2024-07-18:
On my smalltv-ultra and I can't upload gifs.

When I try to upload gifs with 1MB or more, smalltv-ultra crashes, and the upload takes a long time, when finished it shows the file with 0MB and it is not possible to view it.

[image]
[image]
[image]

I pressed to clear the files but it didn't work, I tried with the gifs from the GeekMagicClock/gif/gif 240x240px repository and the same thing happened.
I updated the firmware from V9.0.25 to V9.0.26 but I still have the same error.

[image]

RTormente, 2024-07-18:
I forgot to mention that I also performed the factory reset and besides what I mentioned, I don't know if it's normal for an image to not appear for the weather status.

[image]

GeekMagicClock, 2024-07-19:
Hello,
1. Please check the free space before uploading, your free space shows 479KB, so if you upload more than that, yes it will fail.
2. The gray circle icon on the top right corner means clear at night.
Thank you.

RTormente, 2024-07-20:
I really didn't realize
sorry

closed 2024-07-20 by RTormente