GeekMagicClock/smalltv-ultra #98: Weather screen

ninja1a, 2024-07-30:
Suddenly dropped info on weather page
Gui still works
The code for location shows ok but no weather page info
[image]
Can anyone say what to do
I have updated firmware and still same
I have factory reset
And still same

VaultBoy88, 2024-07-30:
Same on my devices as well. Api change?

vlgrilo, 2024-07-30:
Same on my devices

vlgrilo, 2024-07-30:
Same on my devices
[image]
[image]

sodfy16, 2024-07-31:
Same problem here. I did a packet capture on a device with `Ultra-V9.0.26` firmware;
Two requests are sent - and weather data collected from two _different_ providers:

**Current weather**
`GET http://api.openweathermap.org:80/data/2.5/weather?id={MY_CITY_ID}&appid={MY_API_KEY}&units=metric&lang=en HTTP/1.1\r\n`
The response is a `200 OK` and includes the current weather.

**Forecast (the issue at hand)**
`GET http://api.weatherapi.com:80/v1/forecast.json?key=26934d27353840eb95a93640242505&q=57.01,10.06&days=3&aqi=no&alerts=no`
The response is a `403 FORBIDDEN` with reason `/error/message:API key has reached maximum calls per month allowed limit.`

Notice that the key used for the forecast is not our _individual_ Openweathermap one, but a hardcoded key to weatherapi.com.
My guess is that all products use the same key for the forecast - and it is one without a paid plan and thus limited to 1M calls per month ([source](https://www.weatherapi.com/pricing.aspx)).
So, we'll all have our forecasts tomorrow August 1st, and the issue will reappear near the end of each month until fixed.

[image]

GeekMagicClock, 2024-07-31:
> Same problem here. I did a packet capture on a device with `Ultra-V9.0.26` firmware; Two requests are sent - and weather data collected from two _different_ providers:
>
> **Current weather** `GET [...]

Yes, your analysis is correct, but we have paid plans, seems still run out and get 403 response,  Augest 1th will be recover. So I edit the KEY part, please don't post any key here public. And in future versions, we may offer another key input option too.

ninja1a, 2024-08-01:
All mine came back as said

closed 2024-08-02 by GeekMagicClock

Lozman2962, 2025-01-29:
> Again the same problem, the forecast page does not update info, the rest of the weather pages work properly.
> 
> Anyone with the same problem?

Same thing here, since maybe 2 days ago.
I thus made a free account at openweathermap.org. 
Used my personal API key, added it like 15hrs ago, still nothing.
If i make an account at weatherapi.com and input my API key this will not work right? since its hardcoded.
Thanks for anymore help :)

jf