GeekMagicClock/smalltv-ultra #102: Alternative wifi network

Ashwagada, 2024-08-11:
Hello,

Many devices have an alternative wifi option (in case the main network has problems).

I think it would be good if you could implement something like that in SmallTV Ultra.

GeekMagicClock, 2024-08-14:
Hi, I once had the same idea as you, but the actual implementation may lead to some inexplicable problems, and the user does not know what is happening. So I decide not to automatically switch to another WIFI for now.  But it's open to talk.

Ashwagada, 2024-08-14:
Could you implement this for those interested?

For example, we enter the name and password of the alternative network, and the quality of the connection will be checked every minute.

When both networks are active, it connects to the best one.