GeekMagicClock/smalltv-ultra #103: Huge vulnerability - fallback to AP mode (GifTV) !

mnemologic, 2024-08-14:
When any power outage occurs and device can't connect to current WiFi network
(after three attempts) it falls back to GifTV AP Mode that anyone can connect to
(Open WiFi, no password protected).

The problem is that device configuration site remembers previous settings
(proper WiFi SSID, password, OpenWeather API Key, etc.)
And everything is in plain text exposing entire WiFi network credentials!

So in such circumstances practically anyone, with no special IT skills needed,
can connect to GifTV hotspot and see your current wifi password.

GeekMagicClock, 2024-08-15:
Hi, that is a strategy for users to re-configure it, actually in recent production version, the password is hidden on the web.

mnemologic, 2024-08-15:
They used to say the same over two decades ago - "It's not a bug, it's a feature!" ;)

Best solution would be:

1. Randomly generated GifTV hotspot password that is required and displayed on device screen to protect from literally anyone.
(or even fixed password to lower the random access risk)

and / or

2. Password protected configuration site to protect from hostile individuals inside your WiFi network.

GeekMagicClock, 2024-08-16:
Your sarcasm makes sense. :)

Although this is a low probability event, I admit that it is not impossible. And your solution is great. I hope to add this function without increasing the user complexity, so let me try something. I might be inclined to still use open WIFI and add a site that randomly generates passwords and the passwords are displayed on the screen. What do you think?

mnemologic, 2024-08-16:
You can leave it "as is" ;)

It's not my project, but in my opinion it would be much more honest
to either provide at least minimum security for GifTV AP Mode
or to inform/warn buyers of such a "feature".

Especially when leaving this useful device turned on unattended for a long time
in a home or office network environment when such a scenario can occur at any moment.

GeekMagicClock, 2024-08-16:
Yeah, thank for the posting.
I just came out another idea, If this device is in AP mode, and no one connected for like 2 minutes, then turn off the AP mode.

Ashwagada, 2024-08-16:
Hello,

The best solution is to generate a random password (numbers, letters, special characters) displayed on the screen & turn off AP mode after a few minutes.

mnemologic, 2024-08-24:
Since no new firmware update and you want to know my opinion:

As simple solution as a one cheap physical tactile button and a bit of code tweaks would solve the issue.
If user wants to enter configuration mode (AP mode) he/she has to hold the button during boot time.
Every other time the device will boot into normal (client) mode.

Many of us (buyers) are able to use soldering iron.
You would just have to explain which GPIO pins we have to solder wires to.

No need for overgrowing amount of source code.
Just fork the code for those who are able to do soldering procedure themselves.

gthb96, 2024-09-04:
Quick and dirty fix in the meantime, just show asterisks but a proper solution would be great afterwards.

JavierCervilla, 2024-10-11:
> They used to say the same over two decades ago - "It's not a bug, it's a feature!" ;)
>
> Best solution would be:
>
> 1. Randomly generated GifTV hotspot password that is required and displayed on device [...]

This is the good solution, also if you want to reduce more user complexity at same time, you can add a QR code with the password so is easier, idk if a qr code algo fits but... just my 2 cents

mnemologic, 2024-11-27:
I'm fed up with thus device.
Not even a bit secure.

Not only when booting up but also temporary wifi connection problems or DeAuth attacks
result in falling back to open AP mode.
I woke up today with device in AP mode despite being wired to a USB power bank acting as a UPS.
Device has been connected to my wifi network for three days straight and suddenly, without any warning, fell into AP mode.

Not a hint of improved firmware since reporting the issue mid-August 2024.
Dangerous as hell!

A cute, eye-catching hardware trojan horse device!

gthb96, 2024-11-27:
You're right.
It's like always: sell cheap device - support it long enough to not cause scepticism at the beginning - forget about it

Rafajel9296, 2024-12-26:
Did the time reset automatically, and now I can't change it, what should I do?

Gunni, 2025-01-18:
Please do not send the wifi password to the browser when browsing the config age.

Whether in AP mode or accessing via lan, just never send the password to the user, only accept POSTS with updates for it.

mnemologic, 2025-01-30:
@GeekMagicClock 

Could you please at least add an option in configuration page (check box)
allowing to switch off automatic falling back to AP Mode?

Since there is a possibility to activate this mode by interrupting boot process twice
any advanced user should be satisfied with at least voluntary deactivating auto AP Mode.

mnemologic, 2025-02-11:
> Please do not send the wifi password to the browser when browsing the config age.
> 
> Whether in AP mode or accessing via lan, just never send the password to the user, only accept POSTS with updates for it.

@GeekMagicClock 
At least this you could do ^^^^^^

Leaving these fields blank every time the browser loads config page.
Fields like SSID, PASSWORD and API_KEY(s)

Instead you could place below short info like - "SSID set", "PASSWORD set", etc.)
or just don't put any info at all.

GeekMagicClock, 2025-02-17:
OK, if you like, I will upload a version here, return the password as "********" instead.

GeekMagicClock, 2025-02-18:
[FW-Smalltv-Ultra-V9.0.32.zip](https://github.com/user-attachments/files/18835427/FW-Smalltv-Ultra-V9.0.32.zip)
In the V9.0.32, the password filed will be fixed return as "****", you need to change this to your correct password before click "SAVE".

mnemologic, 2025-02-18:
> [FW-Smalltv-Ultra-V9.0.32.zip](https://github.com/user-attachments/files/18835427/FW-Smalltv-Ultra-V9.0.32.zip) In the V9.0.32, the password filed will be fixed return as "****", you need to change this to your correct password before click "SAVE".

Thanks for that. I hope it will be implemented in further official versions as well.

There are users (like I am) who prefer what's essential (security, crucial weather info)
over the graphical widgets, animations, colorful images, etc.

mnemologic, 2025-02-19:
>V9.0.32

BTW, in this version
in "Weather Clock Today" mode
current temperature and humidity readings and bars
does not update automatically.

Manually switching to other theme and coming back to "Weather Clock Today"
helps updating these values but they stay not updated afterwards.

In "Simple Weather Clock" temperatur reading auto-updates correctly.

GeekMagicClock, 2025-02-20:
OK, thanks, I'll check it.