GeekMagicClock/smalltv-ultra #107: Update

Ashwagada, 2024-09-20:
Hello,

Can we expect an update anytime soon?

I won't deny that some time has passed since the last version, and bug reports and suggestions are increasing.

lovedeliver, 2024-09-27:
> Model: SmallTV-Ultra,Version: Ultra-V9.0.26 stopped showing weather forecast for 3 days ahead. shows all days 0%, works on the main screen. stopped working a few days ago

Same here.

GeekMagicClock, 2025-02-09:
@napk1nn  Regarding the issue of weather forecast update, I recommend updating to version V9.0.29, allowing users to set up the weather forecast API to avoid the inability to display weather forecast information due to the monthly exhaustion of public free KEYs.

[FW-Smalltv-Ultra-V9.0.29.zip](https://github.com/user-attachments/files/18723496/FW-Smalltv-Ultra-V9.0.29.zip)

closed 2026-02-01 by GeekMagicClock