GeekMagicClock/smalltv-ultra #110: Reboot -> AP mode is a pain i.t.a.

southy-j, 2024-10-18:
Thanks for the great work, this is a lovely little device :-)

 I’m frequently working on my house electrical and thus have frequent power outages.
Turns out my WiFi (which also goes down) is slower to recover than this little device here.
Which  means it goes into AP mode.
This is really a pain.
Can you PLEASE do something about it? 

- give it significantly more time to try reconnect 
- or let it try reconnect once more after 5 min (power up > try to connect > AP mode > 5 min with no client associated > reboot with new attempt to connect)
- Or make this configurable somehow 
- Or: first reboot: try to reconnect for longer time, only after a second reboot go into AP mode

Any other behavior would be much appreciated, this is really annoying.

southy-j, 2024-10-19:
I just Right now noticed the „delay reconnect to WiFi“ option. 
is that new? I wasn’t aware of that option.
 Sorry for my rant, could be this solves the problem… I’ll investigate.

GeekMagicClock, 2025-02-09:
Yes, that's a solution for this situation.

closed 2025-02-09 by GeekMagicClock