GeekMagicClock/smalltv-ultra #113: Won’t connect to WiFi / boot loops

outlierforlife, 2024-10-23:
This is the second smart TV ultra (out of four) I have gotten that will not get past initial setup. I’m literally getting a 50% failure rate from your Ali express store. It just goes back to the initial GIFTV Wi-Fi set up screen, but it still has all of my Wi-Fi information stored.. hit save it restarts and goes back to the gift GIFTV Wi-Fi initial setup screen. I had success entering Seoul as my city on one of the other smart ultras I have (that was also giving me trouble), after completing set up and then changing it to my current city worked. But this one once again, will not get past the initial set up. I have tried every firmware from 9.0.19 and up.. same result.

outlierforlife, 2024-10-23:
I’ve gotten it to connect to my Wi-Fi one time. The settings are stored, But it keeps loading the GIFTV Wi-Fi. Sometimes when I plugged it back in it still has the gift TV screen loaded. And does not load the system again. I have to re-unplug it and plug it back in and it reboots.

GeekMagicClock, 2024-10-25:
Hello, seems like a wifi compatible problem. Have you tried with your phone's 2.4G hotspot? Or could you share your wifi settings of the router?

outlierforlife, 2024-10-27:
But two other ones work. I currently have a small-TV pro and a smallTV-ultra that is currently connected. My network runs in both 2.4 and 5g. It’s not like I’m entering these details and hidden ssid. It is scanning and seeing the Wi-Fi network I’m using. I have already run through the troubleshooting guide. Put it in compatibility mode too. Even tried to have it connect to my cell phone hotspot and a 2.4ghz only hotspot on my laptop. No go.

GeekMagicClock, 2024-10-28:
OK, that should not happen, let's check step by step.
Can you show your details of wifi settings on your router?

outlierforlife, 2024-10-30:
My router is an Askey SAX1V1K (Spectrum’s supplied, wifi6 router). It has the ability to go into 2.4 only mode, but I did not need to do that for any of my pros (which I now have three of) and my other standard weather tv just works no problem.

But even weirder is I just found a way to get it to load up every time.. it has my settings stored for router and weather. I install the firmware for small TV main v3.0.33,  it boots up to the GIFTV screen, I then reconnect and install the current firmware for the small TV ultra. And It loads. If I unplug it and plug it back in, it will go right back to the GIFTV screen. But I repeat the firmware upgrade process, and it goes back to the correct weather screen. I really can’t explain this lol but it’s working kind of

GeekMagicClock, 2024-10-30:
This strange phenomenon may be related to this trick. When the startup progress bar just starts to run, the power is cut off twice, and the third time it is turned on, it will directly enter the GIFTV hotspot mode.

The pro has different better cpu and software, that may be the reason it works good, wifi compatiblity is better.

For the ultra, please check if the 2.4g settings on the router support 802.11b/g/n mode, or you could try switch the modes to check if this will work.

soondin, 2024-11-13:
Hello GeekMagic,
My ultra have same problems.
Okay to setup 192.168.1.4 but after save my wifi (2.4G with bgn).
It display New IP but boot roop.

GeekMagicClock, 2024-11-13:
> Hello GeekMagic, My ultra have same problems. Okay to setup 192.168.1.4 but after save my wifi (2.4G with bgn). It display New IP but boot roop.

Hello, is that version V9.0.26? And what is the theme you set after the boot is done? Did you upload your own gif files?

soondin, 2024-11-13:
> 
> Hello, is that version V9.0.26? And what is the theme you set after the boot is done? Did you upload your own gif files?

Version is 9.0.26. I don't changed theme maybe it was default. And I do not upload own gif.
I have deleted all other defaul gifs. but symptoms same.

Rafajel9296, 2024-12-26:
Did the time reset automatically, and now I can't change it, what should I do?

bajinder1010, 2025-01-20:
not able to connect to wifi (2.4G with bgn)

Olxsio, 2025-05-13:
It sounds weird, but *TRUST ME*.
Try a bunch of other cables and power adapters. *trust me, it will eventually work!*

HR4719, 2026-09-26:
I was having the same issue with this clock, I have the basic version, finally figured out if you lower  your Wi-Fi router security from WPA 3, to WPA 2, it will connect to Wi-Fi