GeekMagicClock/smalltv-ultra #116: no funktion weather forecast Model: SmallTV-Ultra,Version: Ultra-V9.0.26

elzershark, 2024-12-29:
no funktion weather forecast Model: SmallTV-Ultra,Version: Ultra-V9.0.26

jesuschristpose, 2024-12-29:
The weather forecast isn't working for me either as of today. The live weather appears to still be working.

Forecast returns 0c for min/max with 0% change of precipitation for the next three days which is incorrect.

jesuschristpose, 2024-12-30:
Can confirm that signing up for a free openweathermap.org account and supplying the API key does not fix this.

jesuschristpose, 2025-01-01:
This has started working again for me again. Perhaps it was an api issue with openweathermap?

pottenmak, 2025-01-28:
Same problem here. If I check with wireshark, it's a problem with the access key : API key has reached maximum calls per month allowed limit.

Is it possible to use a self generated key ?

GET /v1/forecast.json?key=26934d27353840eb95a93640242505&q=51.21,3.22&days=3&aqi=no&alerts=no HTTP/1.1
host:api.weatherapi.com:80

HTTP/1.1 403 Forbidden
Date: Tue, 28 Jan 2025 10:40:11 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: keep-alive
Vary: Accept-Encoding
Server: BunnyCDN-FR1-1187
CDN-PullZone: 93447
CDN-Uid: 8fa3a04a-75d9-4707-8056-b7b33c8ac7fe
CDN-RequestCountryCode: BE
Cache-Control: public, max-age=0
x-weatherapi-qpm-left: 0
CDN-ProxyVer: 1.19
CDN-RequestPullSuccess: True
CDN-RequestPullCode: 403
CDN-CachedAt: 01/28/2025 10:40:11
CDN-EdgeStorageId: 951
CDN-Status: 403
CDN-RequestTime: 0
CDN-RequestId: ae5a5cd91432a506b99a5527201a50bb
CDN-Cache: BYPASS

{"error":{"code":2007,"message":"API key has reached maximum calls per month allowed limit."}}

Huege, 2025-01-29:
Hi, same here, got a fresh API key from OpenWeather and still see no forecast.

Mashryock, 2025-01-29:
Same as others, forecast just showing 32 F for all numbers

Elpi007, 2025-02-01:
Same here. I already had the issue a couple of months ago. Then I sent a mail to the support contact person, and a couple of days after, it was fixed.

But since a couple of days, not working again :/

GeekMagicClock, 2025-02-03:
@pottenmak Hello, please do not expose the public key for reviewing the place. Yes, we plan to offer options for users to set their own keys.

Huege, 2025-02-04:
Update: since yesterday it works again! Nothing changed on settings, fw, etc..

closed 2025-02-04 by elzershark

Mashryock, 2025-02-04:
Works for me now also ater a reboot.  Thanks!

GeekMagicClock, 2025-02-09:
@elzershark @pottenmak @Huege @jesuschristpose @Mashryock  
Regarding the issue of weather forecast update, I recommend updating to version V9.0.29, allowing users to set up the weather forecast API to avoid the inability to display weather forecast information due to the monthly exhaustion of public free KEYs.

[FW-Smalltv-Ultra-V9.0.29.zip](https://github.com/user-attachments/files/18723498/FW-Smalltv-Ultra-V9.0.29.zip)

pottenmak, 2025-02-09:
Thank you, new firmware works like a charm with a self generated key for the forecast

jesuschristpose, 2025-02-09:
Thank you! I also see the gif animation speed is slower. I prefer this.

Elpi007, 2025-02-09:
lol, for me, the fact taht the gif is now slower ... is not good ... my GIF seems to be moving in slow motion now ...

GeekMagicClock, 2025-02-10:
@jesuschristpose  @Elpi007   OK, please check  this V9.0.30
https://github.com/GeekMagicClock/smalltv-ultra/issues/123#issuecomment-2647123271

Elpi007, 2025-02-10:
> [@jesuschristpose](https://github.com/jesuschristpose) [@Elpi007](https://github.com/Elpi007) OK, please check this V9.0.30 [#123 (comment)](https://github.com/GeekMagicClock/smalltv-ultra/issues/123#issuecomment-2647123271)

Thanks, th eprolem with the Gif has now been fixed (thanks for the quick fix).

But unfortunately, a new issue has appeared. When the devie is turned ON (or rebooted) the time is not correct.
There is a difference of 11 hours (in my case, I am based in Germany, and it shows 00:37 instead of 11:37 for example).

But if I save a setting from the setting console, then the correct time is being showed again.