GeekMagicClock/smalltv-ultra #118: Scrolling text for long city names

thisdev, 2025-01-03:
If you live in a city with a long city name, the text breaks and disappears behind the IP address. For long names (greater than x characters), you could use a running text instead of a fixed text. See example below:
[image]

GeekMagicClock, 2025-01-06:
Yeah, I was thinking of add a "Short name" option to allow people shorten the long city name, is that ok too?

thisdev, 2025-01-06:
> Yeah, I was thinking of add a "Short name" option to allow people shorten the long city name, is that ok too?

This could also be a nice solution, sure.

GeekMagicClock, 2025-02-10:
@thisdev Yes, we could.