GeekMagicClock/smalltv-ultra #122: request for setting up a theme change schedule

Bigbadare, 2025-02-08:
I have several boxes at home, mainly in the bedrooms (5), for me and my children.

I would like to be able to switch from the timeclock2 theme at night to the weather clock today theme during the day without having to connect to the web interface.
Because indeed, at night, for the children and even for me, the timeclock2 theme is very readable at night, and during the day we would like to be able to enjoy another more complete theme like the weather clock today theme.

Thank you!

GeekMagicClock, 2025-02-09:
Hello, thanks for letting me know, I can understand your needs, I was thinking of a flexible setup method that lets people set up themes switch schedule.

Bigbadare, 2025-02-09:
Hello GeekMagicClock,

Thank you very much !

Maybe the solution I had thought of would simply be to integrate the choice of a desired theme in the "night mode" setting, but indeed set up themes switch schedule would be perfect 😊

Sorry for my level of English I use google to translate