GeekMagicClock/smalltv-ultra #123: UPDATE V9.0.29 - GIF Issue In Weather Mode

TwinsMasterP, 2025-02-10:
Hello,

 I just made the new update to V9.0.29 and in weather mode the frame rate of my GIF is very bad compared to V9.0.26. I returned to V9.0.26 and my GIF run correctly now. 

Thanks!

GeekMagicClock, 2025-02-10:
[FW-Smalltv-Ultra-V9.0.30.zip](https://github.com/user-attachments/files/18729501/FW-Smalltv-Ultra-V9.0.30.zip)  

Hi, thank you for the report, please try this one.

TwinsMasterP, 2025-02-10:
Thanks, GIF run correctly now with V9.0.30.

Ashwagada, 2025-02-10:
@GeekMagicClock Hello, now it works fine, but you can see the stop and start. No smooth transition. I didn't notice this problem in earlier versions.

Elpi007, 2025-02-10:
Same here. Gif issue has een solved, but the date and time is wrong on the Weather Clock today.
The Weather Forecast shows the correct time.

You have to modify and save some settings so that the date and time can be shown correctly.

GeekMagicClock, 2025-02-11:
Hi, @Elpi007 @Ashwagada @TwinsMasterP  I tried here, reboot the device, seems all right, time is displayed correctly, but I guess in your network enviroment, time sync from NTP server througth UDP seems blocked, please try the v9.0.31.

[FW-Smalltv-Ultra-V9.0.31.zip](https://github.com/user-attachments/files/18748399/FW-Smalltv-Ultra-V9.0.31.zip)

Elpi007, 2025-02-11:
> Hi, [@Elpi007](https://github.com/Elpi007) [@Ashwagada](https://github.com/Ashwagada) [@TwinsMasterP](https://github.com/TwinsMasterP) I tried here, reboot the device, seems all right, time is [...]

Just installed it, and after the reboot, eveything looks fine now ! Time is correct, GIF moving perfect!
Thanks again for the quick fix !!!

Ashwagada, 2025-02-11:
Time and date now work properly :)

TwinsMasterP, 2025-02-11:
Same for me everything looking fine now.

Thanks!