GeekMagicClock/smalltv-ultra #127: Boot loops with V9.0.31.bin

ultraboost, 2025-02-27:
I updated two units from V9.0.26.bin to V9.0.31.bin. Both booted up then after 10 seconds rebooted again and again. I tried a factory reset which didn't help. I then put V9.0.26 back on both units and they stopped rebooting and are stable.

frosch90, 2025-02-28:
V9.0.26.bin   :  weatehr forecast   not suppert

  V9.0.31.bin   reboot  roop

GeekMagicClock, 2025-03-01:
I guesss, you may not follow the steps to set weather forecast key in version V9.0.31, please try check the instructions first.

ultraboost, 2025-03-01:
> I guesss, you may not follow the steps to set weather forecast key in version V9.0.31, please try check the instructions first.

This is about the unit rebooting constantly, not setting weather forecast keys.

GeekMagicClock, 2025-03-02:
Did you changed the weather forecast key? If not, then that's not ralated, but if yes, that's possible.

ultraboost, 2025-03-02:
> Did you changed the weather forecast key? If not, then that's not ralated, but if yes, that's possible.

No it was not possible to even try as it reboots within 10 seconds.

GeekMagicClock, 2025-03-03:
> > Did you changed the weather forecast key? If not, then that's not ralated, but if yes, that's possible.
> 
> No it was not possible to even try as it reboots within 10 seconds.

So you just try the update from V9.0.26 to V9.0.31 and then the reboot loop happens, I'll try that.

ultraboost, 2025-03-03:
> > > Did you changed the weather forecast key? If not, then that's not ralated, but if yes, that's possible.
> > 
> > 
> > No it was not possible to even try as it reboots within 10 seconds.
> 
> So you just try the update from V9.0.26 to V9.0.31 and then the reboot loop happens, I'll try that.

Yes on two units. I tried upgrading and downgrading both twice and both times v31 was unstable.

frosch90, 2025-03-04:
p{margin-top:0;margin-bottom:0}
    

    

    
    
        
        대용량파일 1개 (59.84MB) ~ 2025.04.03 (30일 보관, 100회 다운로드 가능)
        
    
    
        
    
    
    
        
            
                
                
                    
                        
                    
                    
                    
                        
                    
                    
                    VID_20250304_090628.mp4  (59.84MB)
                
                
            
        
    
    
    
        
    
    
        
            
        
    
    
        
    
    

       Hi          So late , today comeback.          Did you changed the weather forecast key?         ->  NO.  default           today check  ,  forecast  Good !,  The weather in the future is good.        Version  9.0.31  reboot loop   :(         Good LUCK  !--------- 원본 메일 ---------

  
    보낸사람: ultraboost ***@***.***>
    받는사람: GeekMagicClock/smalltv-ultra ***@***.***>
      참조: frosch90 ***@***.***>,Comment ***@***.***>
    날짜: 25.03.03 22:22 GMT +0900
    
    제목: Re: [GeekMagicClock/smalltv-ultra] Boot loops with V9.0.31.bin (Issue #127)
    
    
  
  
    
  

Did you changed the weather forecast key? If not, then that's not ralated, but if yes, that's possible.

No it was not possible to even try as it reboots within 10 seconds.

So you just try the update from V9.0.26 to V9.0.31 and then the reboot loop happens, I'll try that.

Yes on two units. I tried upgrading and downgrading both twice and both times v31 was unstable.

Did you changed the weather forecast key? If not, then that's not ralated, but if yes, that's possible.

No it was not possible to even try as it reboots within 10 seconds.

So you just try the update from V9.0.26 to V9.0.31 and then the reboot loop happens, I'll try that.

Yes on two units. I tried upgrading and downgrading both twice and both times v31 was unstable.

closed 2025-03-27 by ultraboost