GeekMagicClock/smalltv-ultra #130: The weather forecast is only for tomorrow

frosch90, 2025-03-06:
[image]

HI.

 The weather forecast is only for tomorrow.   ;( 

 Ver   V9.0.26 
 Weather API KEY is default.

GeekMagicClock, 2025-03-07:
Hi, V9.0.31 recommended, and set your own api key.

frosch90, 2025-03-08:
Hi!

Ver  V9.0.31  Update

 default KEY   -    choose theme   1.  weather clock today  - OK ..  GOOD
                                                      2. weater forecast      - NOT BAD     -  reboot  repeat
                                                      
 My KEY         -      choose theme   1.  weather clock today  - OK ..  GOOD
                                                       2. weater forecast      - NOT BAD     -  reboot  repeat

frosch90, 2025-03-08:
^^

cdelavega950, 2025-03-09:
with Ver V9.0.26.
At this moment it has just been fixed and the forecast for three days is already out.
with V9.0.31 and my own API key, I have not known how to configure it and make it work

butinhi, 2025-03-10:
Me tooo, Reboot when choose weather forecast

GeekMagicClock, 2025-03-12:
Hi, the weather forecast key is different from weather today's key, please check the instructions in the forecast key input area.

frosch90, 2025-03-12:
Weather API Key
Weather Forecast API Key

Use two different keys.

One of the two devices was confirmed to be a device problem and was returned.

It's going well now.

butinhi, 2025-03-12:
> Hi, the weather forecast key is different from weather today's key, please check the instructions in the forecast key input area.

I used the instructions but not working. Always reboot and not show forcecast weather