GeekMagicClock/smalltv-ultra #131: Can I update?

tomektarnowski, 2025-03-10:
Hi team, not sure if that is translation issue, but, can I update `Model: SmallTV-Ultra,Version: Ultra-V9.0.26` by using [Ultra-V9.0.31](https://github.com/GeekMagicClock/smalltv-ultra/tree/main/Ultra-V9.0.31) FW, or is it a different HW?

ultraboost, 2025-03-19:
I would not bother. v31 has a problem that causes continuous rebooting.

GeekMagicClock, 2025-03-23:
@tomektarnowski Hi, could be updated, try the zip format firmware file if you failed.

GeekMagicClock, 2025-03-23:
@ultraboost  Hello, did you set your own weather forecast key but did not follow the full instructions ?

ultraboost, 2025-03-23:
> [@ultraboost](https://github.com/ultraboost) Hello, did you set your own weather forecast key but did not follow the full instructions ?

It's not possible to set my own weather forecast key as it reboots within 10 seconds over and over

ultraboost, 2025-03-23:
> [@tomektarnowski](https://github.com/tomektarnowski) Hi, could be updated, try the zip format firmware file if you failed.

I've tried both

GeekMagicClock, 2025-03-23:
> > [@ultraboost](https://github.com/ultraboost) Hello, did you set your own weather forecast key but did not follow the full instructions ?
> 
> It's not possible to set my own weather forecast key as it reboots within 10 seconds over and over. Tried it on two units and both the same.

OK, then, please follow the steps about "Factory Reset" method in the manual.

ultraboost, 2025-03-23:
> OK, then, please follow the steps about "Factory Reset" method in the manual.

I already tried a factory reset on both units and that didn't help. Still rebooting every 10 seconds with v31.

CpUofFlINe, 2025-03-26:
hello , I just bought this device and ofc i made v31 update 
and yes it reboot 
i solve this way : on the weather page (where i got key , go to API and uncheck everything except : 

[image]

now it works normal

GeekMagicClock, 2025-03-27:
> hello , I just bought this device and ofc i update to v31 update and yes it reboot i solve this way : on the weather page (where i got key , go to API and uncheck everything except :
> 
> [image]
> 
> now it works normal

Yes, that's correct!!! As is descripted in the instructions of setting the weather forecast key!

GeekMagicClock, 2025-03-27:
@ultraboost  I doubt if you have follow the instructions of "Factory Reset" and the setting of set the "Forecast API Key", as mentioned in the instructions, "Step 3: [ IMPORTANT ]
Please strictly follow the following option settings, otherwise it may cause the device to restart frequently because of limited ram space!."

ultraboost, 2025-03-27:
> [@ultraboost](https://github.com/ultraboost) I doubt if you have follow the instructions of "Factory Reset" and the setting of set the "Forecast API Key", as mentioned in the instructions, "Step 3: [...]

I completely reset both my units back to factory settings. 

Once v31 is on I barely have time to so anything before it reboots. I don't have time to set the forecast API key or modify settings before it reboots.

ultraboost, 2025-03-27:
> hello , I just bought this device and ofc i update to v31 update and yes it reboot i solve this way : on the weather page (where i got key , go to API and uncheck everything except :
> 
> [image]
> 
> now it works normal

This has fixed it. Thankyou!!!! 
Problem is, in the directions the jpeg is so small it's impossible to read without microscopes for eyes!

[image]

GeekMagicClock, 2025-03-28:
OK, I have updated the images In the instructions.

cdelavega950, 2025-03-28:
After generating my own key with only the 5 parameters indicated in the instructions, I tried to save the new forecast API key but it is not saved in the configuration.

[image]

ultraboost, 2025-03-28:
> After generating my own key with only the 5 parameters indicated in the instructions, I tried to save the new forecast API key but it is not saved in the configuration.
> 
> [image]

I noticed that too. It looks like it's not saved but it has. Check the website for number of calls and you should see it made some. 

Maybe this can be fixed in a new version?

cdelavega950, 2025-03-28:
You're right. The API statistics indicate that it works, although it's not shown in the configuration.

GeekMagicClock, 2025-04-02:
The key will not be seen from the web for secure reason.

ultraboost, 2025-04-02:
> The key will not be seen from the web for secure reason.

The weather API key above (4) is visible though. Should it not be?

Feika1981, 2025-06-24:
Where did you even find these checkboxes? I registered on openweathermap.org it just creates an api key without any settings and the clock goes into reboot all the time?

GeekMagicClock, 2026-02-01:
@controlrepo  How about V9.0.43?

Feika1981, 2026-02-05:
I installed the latest firmware yesterday, but they still reboot as soon as they connect to Wi-Fi.